maakbaas/esp8266-iot-framework #135: Fetch example throws exception

rdcavanha, 2021-12-25:
Hello,

When running the [fetch example](https://github.com/maakbaas/esp8266-iot-framework/blob/master/examples/fetch/fetchExample.cpp) on my nodemcu, it throws an exception and restarts itself (see monitor output below).

I have tried rebuilding the certificate store  with `-DREBUILD_CERTS`, but nothing changes. Also, I'm just running the `fetch` example, nothing else. I suspect that the issue might be related to memory exhaustion. Any ideas?

Note: If I use this other example:
[BasicHttpsClient.ino](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266HTTPClient/examples/BasicHttpsClient/BasicHttpsClient.ino), it works perfectly fine and the HTTPS request succeeds if I set `client->setInsecure();`. I am aware that using `insecure` is not the ideal solution, hence why I would like to use the framework's `fetch` module.

Thank you in advance and happy holidays!

```
Exception (29):
epc1=0x4000df64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: sys
sp: 3fffec10 end: 3fffffb0 offset: 0190
3fffeda0:  00000005 3fff0140 00000002 401004c0  
3fffedb0:  4027cbf3 3fff2b2c 00000005 4027cb88  
3fffedc0:  00000002 4027cb2f 00000002 4027bc88  
3fffedd0:  4027bcb1 3fffee80 3fff0140 00000016  
3fffede0:  40279718 3fffee80 3ffeffe8 3ffef9d4  
3fffedf0:  3ffec570 3fffee80 3fffee80 40105c39  
3fffee00:  6574754e 20616c6c 47342e32 00000100  
3fffee10:  3ffebedc 7fffffff 00002200 00000001  
3fffee20:  40275a79 00000080 00000006 00000000  
3fffee30:  ffffffe5 3ffefafc 3ffec580 3fff0140  
3fffee40:  3ffef0e0 00000045 00000000 4027a413  
3fffee50:  00000000 3fff2b2c ffffffe5 00000000  
3fffee60:  00000000 3fff0140 00000017 0000000f  
3fffee70:  40000f58 00000000 b670b0c3 00000000  
3fffee80:  00000000 11110909 00640104 0000004b  
3fffee90:  3ffec594 000000c3 3ffec5af 3ffec588  
3fffeea0:  3ffec5b5 3ffec594 3ffec5a2 3ffec5c7  
3fffeeb0:  00000000 00000000 3ffec5cd 3ffec63d  
3fffeec0:  3ffec5f4 3ffec610 00000000 00000000  
3fffeed0:  00000000 00000000 00000045 00000000  
3fffeee0:  3fff2dd4 40279e82 3ffef0e0 3fff2b2c  
3fffeef0:  00000000 3fff0140 3ffef0e0 3ffec570  
3fffef00:  3ffec570 000000eb 00000000 00000045  
3fffef10:  00000000 3ffec57a 4028d55b 3ffef0e0  
3fffef20:  3ffec564 3fffdcc0 3ffeb6d8 3ffeb6d8  
3fffef30:  00000080 3ffef0e0 00000000 3ffe88f4  
3fffef40:  4028ce1f 3fffdab0 00000000 4021f626  
3fffef50:  3ffeb6d8 40000f49 3fffdab0 40000f49  
3fffef60:  40000e19 000efed5 00000000 00000005  
3fffef70:  00000000 aa55aa55 000000bd 401054a1  
3fffef80:  401054a7 00000000 00000005 40100af0  
3fffef90:  4010000d 2928657a 000efed5 401000ac  
3fffefa0:  40281d78 3fffef3c 40281d29 3ffffd88  
3fffefb0:  3fffffc0 00000000 00000000 feefeffe  
3ffff710:  feefeffe feefeffe feefeffe feefeffe 
... (a bunch more of feefeffe feefeffe feefeffe feefeffe)
3ffff720:  00000002 00000000 00000020 401002bc  
3ffff730:  40102539 feefeffe 00000002 40102108  
3ffff740:  3ffeb652 40105387 3ffeedc0 fS
```

maakbaas, 2021-12-27:
Doing HTTPS requests can be challenging for the ESP8266 memory, however, the unmodified Fetch example should work. Does it restart immediately on the first attempt, or only after a certain number of requests?

I did a test with the current repository version on a NodeMCU, and I don't have any issues. Therefore I am afraid it will be hard for me to debug this.

You could try if this leads to some more insights:
https://github.com/me-no-dev/EspExceptionDecoder

closed 2021-12-27 by maakbaas

rdcavanha, 2021-12-27:
Thank you, @maakbaas 
I'm starting to think that my board is defective. I will try the same code on another nodemcu that I have.
For now, I would say the issue can be closed since it's not replicable.