maakbaas/esp8266-iot-framework #137: #include "LittleFS.h"

BaumKL, 2022-01-07:
When I try to run the PlatformIO: Build  I get this error message:

Looking for ESPAsyncWebServer.h dependency? Check our library registry!
.pio/libdeps/d1_mini/ESP8266 IoT Framework/src/updater.cpp:3:10: fatal error: LittleFS.h: No such file or directory

******************************************************************
* Looking for LittleFS.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:LittleFS.h"
* Web  > https://platformio.org/lib/search?query=header:LittleFS.h
*
******************************************************************

    3 | #include "LittleFS.h"
      |          ^~~~~~~~~~~~
compilation terminated.
*** [.pio/build/d1_mini/lib193/ESP8266 IoT Framework/updater.cpp.o] Error 1

maakbaas, 2022-01-08:
Do you use the framework as a PlatformIO library, or have you checked out this repository directly?

Can you share your `platformio.ini` file?

BaumKL, 2022-01-08:
Today I reinstalled all of the libraries and then it works

Sorry 😂

closed 2022-01-08 by BaumKL