maakbaas/esp8266-iot-framework #138: ESPAsyncWebServer.h not found

llpamies, 2022-01-11:
I am getting this error:
```
In file included from .pio/libdeps/emonpub/ESP8266 IoT Framework/src/dashboard.cpp:1:
.pio/libdeps/emonpub/ESP8266 IoT Framework/src/dashboard.h:4:10: fatal error: ESPAsyncWebServer.h: No such file or directory
```

maakbaas, 2022-01-11:
Probably an issue with your platformio.ini.

Do you use the framework as a library, or did you check out the repo directly?

Can you share your platformio.ini file?

llpamies, 2022-01-11:
That is my current platformio.ini:
```
[platformio]
default_envs = emonpub

[common]
build_flags = -std=c++17
monitor_speed = 115200
lib_deps_external =
  ESP8266 IoT Framework
  ESPAsyncWebServer
  https://github.com/bblanchon/ArduinoJson

[env:emonpub]
platform = espressif8266
framework = arduino
board = nodemcuv2
build_flags = ${common.build_flags}
lib_deps = ${common.lib_deps_external}
monitor_speed = ${common.monitor_speed}
```

llpamies, 2022-01-12:
I cleaned the local .pio directory, and I added the following deps explicitly:
`lib_deps = ESP8266 IoT Framework, ESP Async WebServer, LittleFS`
and now it works!

closed 2022-01-12 by maakbaas