maakbaas/esp8266-iot-framework #141: Adding "type": "password" in configuration

Eisbaeeer, 2022-01-26:
It will be more secure, if the password is hidden in configuration in case of passwords are used in text fields.
Or is there an other way to asterisk the text field?
Thank you.

maakbaas, 2022-01-26:
Agreed. That one should be easy enough for you to put in a pull request :).

Eisbaeeer, 2022-01-26:
Ahh, Ok. I will try my best ;-)

Eisbaeeer, 2022-01-26:
There is nothing to do ;-) 
It is already usable with "control" like this:
`{
        "name": "password",
        "label": "Password",
        "type": "char",
        "length": 20,
        "value": "blabla",
        "control": "password"
    }`

maakbaas, 2022-01-26:
Ok, I think that is an unintended functionality 😎, I'll look into this later

maakbaas, 2022-09-04:
Addressed in c14776636073471e79461854988e1e522b0904a7

closed 2022-09-04 by maakbaas