maakbaas/esp8266-iot-framework #145: Event on Dashboard Change

marcolino7, 2022-02-18:
Hi,
I am developing code for a Big 7 Segment Clock. I need to change some parameter with sliders and combos, like brighness and color.
Fom Configuration, this is not possible, so I have to use Dashboard, but I need to store values into configurations.
I would like to know if there is a way to start aroutine when a value on dashboard is chaged, like for `configManager.setConfigSaveCallback(saveCallback);`
in order to run code to save dashboard settings to config manager.

Thanks in advance

PS a color picker would be great

Marco

maakbaas, 2022-02-19:
Hi Marco,

A color picker is already possible. Please refer to the documentation in this section:
https://github.com/maakbaas/esp8266-iot-framework/blob/master/docs/config-manager.md#code-generation

Any reason why sliders and combo boxes do not work on the configuration page for you? That should also be possible.

With regards to the callback. This is currently not part of the framework. I don't see another way than storing a copy and polling in your main loop for changes for a particular variable. That should be fairly easy to implement in your own application. I hope in that way you can still achieve what you are looking for.

marcolino7, 2022-02-19:
Hi @maakbaas 
I have an old version of docs on my disk, I just found coloepicker and slider in cofig.
I have an issue with colorpicker I think.
I used this syntax in cofiguration.json
`{
        "name": "dispayColor",
        "label": "Display Color",
        "type": "uint16_t",
        "control" : "color",
        "value": [255, 0, 0]
    },`
But when I compile I got this error in compiled config.cpp:
`.pio\libdeps\esp12e\ESP8266 IoT Framework\src\generated\config.cpp:12:3: error: expected identifier before numeric constant
   12 |  [255, 0, 0],
      |   ^~~
.pio\libdeps\esp12e\ESP8266 IoT Framework\src\generated\config.cpp:12:6: error: expected ']' before ',' token
   12 |  [255, 0, 0],
      |      ^
      |      ]
.pio\libdeps\esp12e\ESP8266 IoT Framework\src\generated\config.cpp: In lambda function:
.pio\libdeps\esp12e\ESP8266 IoT Framework\src\generated\config.cpp:12:6: error: expected '{' before ',' token
.pio\libdeps\esp12e\ESP8266 IoT Framework\src\generated\config.cpp: At global scope:
.pio\libdeps\esp12e\ESP8266 IoT Framework\src\generated\config.cpp:12:12: error: expected '}' before ']' token
   12 |  [255, 0, 0],
      |            ^
.pio\libdeps\esp12e\ESP8266 IoT Framework\src\generated\config.cpp:7:1: note: to match this '{'
    7 | {
      | ^
.pio\libdeps\esp12e\ESP8266 IoT Framework\src\generated\config.cpp:12:12: error: too many initializers for 'const configData'
   12 |  [255, 0, 0],
      |            ^
.pio\libdeps\esp12e\ESP8266 IoT Framework\src\generated\config.cpp:12:12: error: expected ',' or ';' before ']' token
.pio\libdeps\esp12e\ESP8266 IoT Framework\src\generated\config.cpp:14:1: error: expected declaration before '}' token
   14 | };
      | ^
*** [.pio\build\esp12e\libc5e\ESP8266 IoT Framework\generated\config.cpp.o] Error 1`

and opening file, spell checking show an error:
[image]

Do I miss something?

Thanks

maakbaas, 2022-02-20:
I am away from home, so will only be able to look at this further second half of March. Sorry about that.

If anybody else can test the same code in the meantime and report back that would be helpful.

marcolino7, 2022-02-20:
Hi @maakbaas ,
thanks for answer. Perhaps it could be better change subject of the issue or open a new one, with subject more related to the issue.
Let me know.

Marco

Eisbaeeer, 2022-02-21:
Hi @marcolino7 You have to define the colorpicker in "configuration.json" like this:

 `{
        "name": "colorGreen",
        "label": "Color green",
        "type": "color",
        "value": [0,255,0]
    },`

and the folder of "configuration.json" in "platformio.ini" like this:

`build_flags = -DCONFIG_PATH=src/configuration.json`

Then put the configuration in your src path. See example in documentation.
Greets Lars

marcolino7, 2022-02-21:
Hi @Eisbaeeer ,
I defined configuration.json like this:

```
{
        "name": "dispayColor",
        "label": "Display Color",
        "type": "color",
        "value": "[255, 0, 0]"
    },
```

And compiling i got this error:

```
.pio\libdeps\esp12e\ESP8266 IoT Framework\src\generated\config.cpp:12:4: error: expected identifier before ',' token
   12 |  {[,2,5},
      |    ^
.pio\libdeps\esp12e\ESP8266 IoT Framework\src\generated\config.cpp:12:4: error: expected ']' before ',' token
   12 |  {[,2,5},
      |    ^
      |    ]
.pio\libdeps\esp12e\ESP8266 IoT Framework\src\generated\config.cpp: In lambda function:
.pio\libdeps\esp12e\ESP8266 IoT Framework\src\generated\config.cpp:12:4: error: expected '{' before ',' token
.pio\libdeps\esp12e\ESP8266 IoT Framework\src\generated\config.cpp: At global scope:
.pio\libdeps\esp12e\ESP8266 IoT Framework\src\generated\config.cpp:14:1: error: invalid user-defined conversion from '<lambda()>' to 'uint8_t' {aka 'unsigned char'} [-fpermissive]    
   14 | };
      | ^
.pio\libdeps\esp12e\ESP8266 IoT Framework\src\generated\config.cpp:12:3: note: candidate is: 'constexpr<lambda()>::operator void (*)()() const' (near match)
   12 |  {[,2,5},
      |   ^
.pio\libdeps\esp12e\ESP8266 IoT Framework\src\generated\config.cpp:12:3: note:   no known conversion from 'void (*)()' to 'uint8_t' {aka 'unsigned char'}
```

Looking at generated config.cpp
[image]

there is some issue revealed bt spell checker.
Any ideas?

Thanks

Eisbaeeer, 2022-02-21:
Can you paste the whole "configuration.json", plese?
I think there is a problem.

marcolino7, 2022-02-21:
Here it is:

```
[
    {
        "name": "projectName",
        "label": "Project Name",
        "type": "char",
        "length": 32,
        "value": "Big Clock by Pozzy"
    },
    {
        "name": "autoBrig",
        "label": "Automatic Brightness",
        "type": "bool",
        "value": true
    },
    {
        "name": "minBrig",
        "label": "Min Brightness",
        "type": "uint16_t",
        "control": "slider",
        "min": 1,
        "max": 250,
        "step": 1,
        "value": 10
    },
    {
        "name": "maxBrig",
        "label": "Max Brightness",
        "type": "uint16_t",
        "control": "slider",
        "min": 10,
        "max": 255,
        "step": 1,
        "value": 255
    },
    {
        "name": "dispayColor",
        "label": "Display Color",
        "type": "color",
        "value": [255,0,0]
    },
    {
        "name": "showTemp",
        "label": "Show Temperature",
        "type": "bool",
        "value": true
    }
]
```

Eisbaeeer, 2022-02-22:
Ok, the only difference I saw are the "values". I define value like this:
`{
        "name": "WS28_Intensity",
        "label": "LED Stripe brightness (0-255)",
        "type": "uint16_t",
        "control": "slider",
        "min": 0,
        "max": 255,
        "length": 5,
        "value": "128"
    },`

marcolino7, 2022-02-24:
Hi,
i delete .pio folder, and after compiling again it worked.

Regards

Marco

Eisbaeeer, 2022-03-01:
@marcolino7 Then please close the issue.

marcolino7, 2022-03-01:
@Eisbaeeer 
yes, you are right and sorry for delay closing this

closed 2022-03-01 by marcolino7