maakbaas/esp8266-iot-framework #148: Enhancement: Adding "label" in dashboard

Eisbaeeer, 2022-02-27:
Nice to have is a "label" field in dashboard. I tried it, but the framework will be getting unstable, if "label" will be used in dashboard.

@maakbaas : Your framework is more than great! You are the only one who created such like this. Many thanks for your work!
I hope really to get it running on ESP32, too.
The separator and title is an other thing who will be great for that framework. I think it will be easy for you to realize. I wish I´m better in react.
Greets Eisbaeeer

maakbaas, 2022-09-04:
The fields below can be used for purposes of the HTML page (both in configuration.json and dashboard.json) and will have no functional impact:

```json
{
    "type": "separator"
},
{
    "type": "header",
    "text": "Header text here"
},
{
    "type": "label",
    "text": "Label text here"
},
```

closed 2022-09-04 by maakbaas

Eisbaeeer, 2022-09-08:
Tried, but does not work. Also set '-DREBUILD_HTML'

maakbaas, 2022-09-09:
What do you run into? You also set `-DREBUILD_DASHBOARD` or `-DREBUILD_CONFIG`?

Eisbaeeer, 2022-09-09:
You saved my day! -DREBUILD_DASHBOARD and -DREBUILD_CONFIG do the job. Now it is working!