maakbaas/esp8266-iot-framework #154: Failing at Step 9.

Tinteex, 2022-08-16:
Hello, 

i added "lib_deps = ESP8266 IoT Framework" to the platformio.ini and when i hit build the following error appears : 

Processing d1_mini (platform: espressif8266; board: d1_mini; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 (4.0.1) > WeMos D1 R2 and mini
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 @ 3.30002.0 (3.0.2)
 - tool-esptool @ 1.413.0 (4.13)
 - tool-esptoolpy @ 1.30000.201119 (3.0.0)
 - toolchain-xtensa @ 2.100300.210717 (10.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 37 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESP8266 IoT Framework @ 1.10.0
|   |-- ArduinoJson @ 6.19.4
|   |-- ESP8266WiFi @ 1.0
|   |-- ESP8266HTTPClient @ 1.2
|   |   |-- ESP8266WiFi @ 1.0
|   |-- DNSServer @ 1.1.1
|   |   |-- ESP8266WiFi @ 1.0
|   |-- EEPROM @ 1.0
Building in release mode
Compiling .pio\build\d1_mini\lib7a1\ESP8266 IoT Framework\dashboard.cpp.o
Compiling .pio\build\d1_mini\lib7a1\ESP8266 IoT Framework\updater.cpp.o
In file included from .pio\libdeps\d1_mini\ESP8266 IoT Framework\src\dashboard.cpp:1:
.pio\libdeps\d1_mini\ESP8266 IoT Framework\src\dashboard.h:4:10: fatal error: ESPAsyncWebServer.h: No such file or directory

***************************************************************************
* Looking for ESPAsyncWebServer.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ESPAsyncWebServer.h"
* Web  > https://registry.platformio.org/search?q=header:ESPAsyncWebServer.h
*
***************************************************************************

    4 | #include <ESPAsyncWebServer.h>
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
Compiling .pio\build\d1_mini\lib7a1\ESP8266 IoT Framework\webServer.cpp.o
Compiling .pio\build\d1_mini\src\main.cpp.o
In file included from .pio\libdeps\d1_mini\ESP8266 IoT Framework\src\webServer.cpp:1:
.pio\libdeps\d1_mini\ESP8266 IoT Framework\src\webServer.h:4:10: fatal error: ESPAsyncWebServer.h: No such file or directory

***************************************************************************
* Looking for ESPAsyncWebServer.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ESPAsyncWebServer.h"
* Web  > https://registry.platformio.org/search?q=header:ESPAsyncWebServer.h
*
***************************************************************************

    4 | #include <ESPAsyncWebServer.h>
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
.pio\libdeps\d1_mini\ESP8266 IoT Framework\src\updater.cpp:3:10: fatal error: LittleFS.h: No such file or directory

******************************************************************
* Looking for LittleFS.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:LittleFS.h"
* Web  > https://registry.platformio.org/search?q=header:LittleFS.h
*
******************************************************************

    3 | #include "LittleFS.h"
      |          ^~~~~~~~~~~~
compilation terminated.
Compiling .pio\build\d1_mini\FrameworkArduino\spiffs\spiffs_check.cpp.o
Compiling .pio\build\d1_mini\FrameworkArduino\spiffs\spiffs_gc.cpp.o
Compiling .pio\build\d1_mini\FrameworkArduino\spiffs\spiffs_hydrogen.cpp.o
Compiling .pio\build\d1_mini\FrameworkArduino\spiffs\spiffs_nucleus.cpp.o
Compiling .pio\build\d1_mini\FrameworkArduino\spiffs_api.cpp.o
Compiling .pio\build\d1_mini\FrameworkArduino\sqrt32.cpp.o
Compiling .pio\build\d1_mini\FrameworkArduino\stdlib_noniso.cpp.o
Compiling .pio\build\d1_mini\FrameworkArduino\time.cpp.o
Compiling .pio\build\d1_mini\FrameworkArduino\uart.cpp.o
Compiling .pio\build\d1_mini\FrameworkArduino\umm_malloc\umm_info.c.o
Compiling .pio\build\d1_mini\FrameworkArduino\umm_malloc\umm_integrity.c.o
Compiling .pio\build\d1_mini\FrameworkArduino\umm_malloc\umm_local.c.o
Compiling .pio\build\d1_mini\FrameworkArduino\umm_malloc\umm_malloc.cpp.o
Compiling .pio\build\d1_mini\FrameworkArduino\umm_malloc\umm_poison.c.o
*** [.pio\build\d1_mini\lib7a1\ESP8266 IoT Framework\dashboard.cpp.o] Error 1
*** [.pio\build\d1_mini\lib7a1\ESP8266 IoT Framework\webServer.cpp.o] Error 1
*** [.pio\build\d1_mini\lib7a1\ESP8266 IoT Framework\updater.cpp.o] Error 1

how can i fix it? everything in the guide worked up to that point. cheers.

Tinteex, 2022-08-16:
noob failed

closed 2022-09-04 by maakbaas

vvilho, 2023-01-04:
I got the same error. One more noob. What did you do? @Tinteex

vvilho, 2023-01-04:
> I got the same error. One more noob. What did you do? @Tinteex

... yeah. Noob here also #138