maakbaas/esp8266-iot-framework #158: Dependency issues in library

walles12, 2022-11-08:
When compiling with these options in the platformio.ini file:
build_flags = -DREBUILD_HTML -DREBUILD_CONFIG -DCONFIG_PATH=configuration_own.json
the script that runs to generate code from config there are dependency issues which I can't manage to resolve.

For instance webpack seems to have been updated, can you please specify exact versions needed for all dependencies?

This is how far I come, libraries needed:
lib_deps = 
    https://github.com/lorol/ESPAsyncWebServer
    ESP8266 IoT Framework
    LittleFS

When I add the build_flags option above I get these errors:
`CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 (4.0.1) > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 @ 3.30002.0 (3.0.2)
 - tool-esptool @ 1.413.0 (4.13)
 - tool-esptoolpy @ 1.30000.201119 (3.0.0)
 - toolchain-xtensa @ 2.100300.210717 (10.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Running npm ci...
npm ERR! code ERESOLVE
npm ERR! ERESOLVE could not resolve
npm ERR! 
npm ERR! While resolving: styled-components@5.1.0
npm ERR! Found: react@16.14.0
npm ERR! node_modules/react
npm ERR!   peer react@"^0.14.0 || ^15.0.0 || ^16.0.0" from mini-create-react-context@0.3.2
npm ERR!   node_modules/mini-create-react-context
npm ERR!     mini-create-react-context@"^0.3.0" from react-router@5.1.2
npm ERR!     node_modules/react-router
npm ERR!       react-router@"5.1.2" from react-router-dom@5.1.2
npm ERR!       node_modules/react-router-dom
npm ERR!         react-router-dom@"^5.1.2" from the root project
npm ERR!   peer react@"*" from react-icons@4.2.0
npm ERR!   node_modules/react-icons
npm ERR!     react-icons@"^4.2.0" from the root project
npm ERR!   5 more (react-motion, react-router, react-router-dom, react-vis, styled-components)
npm ERR!
npm ERR! Could not resolve dependency:
npm ERR! peer react-dom@">= 16.8.0" from styled-components@5.1.0
npm ERR! node_modules/styled-components
npm ERR!   styled-components@"^5.1.0" from the root project
npm ERR!   peer styled-components@">= 2" from babel-plugin-styled-components@1.10.7
npm ERR!   node_modules/babel-plugin-styled-components
npm ERR!     dev babel-plugin-styled-components@"^1.10.7" from the root project
npm ERR!     1 more (styled-components)
npm ERR!   1 more (styled-normalize)
npm ERR!
npm ERR! Conflicting peer dependency: react@18.2.0
npm ERR! node_modules/react
npm ERR!   peer react@"^18.2.0" from react-dom@18.2.0
npm ERR!   node_modules/react-dom
npm ERR!     peer react-dom@">= 16.8.0" from styled-components@5.1.0
npm ERR!     node_modules/styled-components
npm ERR!       styled-components@"^5.1.0" from the root project
npm ERR!       2 more (babel-plugin-styled-components, styled-normalize)
npm ERR!
npm ERR! Fix the upstream dependency conflict, or retry
npm ERR! this command with --force, or --legacy-peer-deps
npm ERR! to accept an incorrect (and potentially broken) dependency resolution.
npm ERR!
npm ERR! See C:\Users\wallentinh.000\AppData\Local\npm-cache\eresolve-report.txt for a full report.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\wallentinh.000\AppData\Local\npm-cache\_logs\2022-11-08T07_35_39_094Z-debug-0.log
Running npx browserslist@latest --update-db...
[... 111 lines cut ...]
Problem with `npm install caniuse-lite` call. Run it manually.
npm ERR! this command with --force, or --legacy-peer-deps
npm ERR! to accept an incorrect (and potentially broken) dependency resolution.
npm ERR!
npm ERR! See C:\Users\wallentinh.000\AppData\Local\npm-cache\eresolve-report.txt for a full report.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\wallentinh.000\AppData\Local\npm-cache\_logs\2022-11-08T07_35_59_240Z-debug-0.log
Running npm run build...

> esp@1.0.0 build
> webpack --mode=production

clean-webpack-plugin: options.output.path not defined. Plugin disabled...
[webpack-cli] TypeError: Cannot read properties of undefined (reading 'tapAsync')
    at C:\Users\wallentinh.000\node_modules\html-webpack-inline-source-plugin\index.js:21:59
    at Hook.eval [as call] (eval at create (C:\Users\wallentinh.000\node_modules\tapable\lib\HookCodeFactory.js:19:10), <anonymous>:102:1)
    at Hook.CALL_DELEGATE [as _call] (C:\Users\wallentinh.000\node_modules\tapable\lib\Hook.js:14:14)
    at Compiler.newCompilation (C:\Users\wallentinh.000\node_modules\webpack\lib\Compiler.js:1122:26)
    at C:\Users\wallentinh.000\node_modules\webpack\lib\Compiler.js:1166:29
    at Hook.eval [as callAsync] (eval at create (C:\Users\wallentinh.000\node_modules\tapable\lib\HookCodeFactory.js:33:10), <anonymous>:6:1)
    at Hook.CALL_ASYNC_DELEGATE [as _callAsync] (C:\Users\wallentinh.000\node_modules\tapable\lib\Hook.js:18:14)
    at Compiler.compile (C:\Users\wallentinh.000\node_modules\webpack\lib\Compiler.js:1161:28)
    at C:\Users\wallentinh.000\node_modules\webpack\lib\Compiler.js:524:12
    at Compiler.readRecords (C:\Users\wallentinh.000\node_modules\webpack\lib\Compiler.js:986:5)`

walles12, 2022-11-08:
This was a duplicate of #152

closed 2022-11-08 by walles12

maakbaas, 2022-11-08:
Thanks for reporting. I will investigate compatibility with newer node versions for a future update.

maakbaas, 2022-11-16:
@walles12 this is fixed now

sejgit, 2022-12-29:
Still an issue.  All worked once I added to my .ini
lib_deps =
         LittleFS
         esphome/ESPAsyncWebServer-esphome
         maakbaas/ESP8266 IoT Framework