maakbaas/esp8266-iot-framework #164: The cipher AES128-SHA is not supported for TLSv1.2.

Eisbaeeer, 2023-05-24:
If I do a get or fetch to e.g. facebook, I got the following error message:

The cipher AES128-SHA is not supported for TLSv1.2.

`[DEBUG] ---begin---
{"error":{"message":"The cipher AES128-SHA is not supported for TLSv1.2.","type":"GraphInsecureException","code":8,"fbtrace_id":"AbKMSqpy07lODte5e9pgftB"}}
[DEBUG] FreeHEAP: 35552`

If I do that in the EDGE Browser, I get the right informations like before with the framework.

You can replicate this with the following get URL:

`https://graph.facebook.com/v17.0/me/accounts?fields=connected_instagram_account,name&access_token=EAALXhfkoVScBACuykeZC8PU2ceHSw5wVXSv280LVLOF5SMWuPEwzBT3isHLZAM1grussvjLkVSU4LAI0jnN7hBPwRvMZBDqseMne0lbPhqeBtvERX6YuCfpJwnxelY5pSl99rR5B0DhpvBsKGFVHSZAdUh8quvRsGgZB0V6xkSj7wzX5WWw58cielroX6J6EZD`

Eisbaeeer, 2023-05-24:
Ok, you are not abel to replicate, because the accesstoken is now blocked, because I posted it here.  ;-)

Eisbaeeer, 2023-05-26:
Now I tried to set the ciphers in fetch.cpp with following ciphers. But if I set this ciphers, the fetch result is null.
I also tried to comment out the other things like "setInsecure, setSSLVersion" and so on. If I comment out the SetCiphers() fetch to goolge work fine, but fetch to graph.facebook.com running into the error `The cipher AES128-SHA is not supported for TLSv1.2.`
I tried some other cipher´s, too with same result. Is there something else to set to get it running?

`httpsClient->setCertStore(&certStore);

        std::vector<uint16_t> myCiphersList = { BR_TLS_ECDHE_ECDSA_WITH_AES_128_GCM_SHA256, BR_TLS_ECDHE_RSA_WITH_AES_128_GCM_SHA256 };
        httpsClient->setCiphers(myCiphersList);
        httpsClient->setInsecure();
        httpsClient->setSSLVersion(BR_TLS12,BR_TLS12); `

Any idea or help?
Thank you!

Eisbaeeer, 2023-05-29:
Ok, the reason, why it´s broken:
`Encryption
When connecting to our servers your client must use TLS and be able to verify a certificate signed using [sha256WithRSAEncryption](http://www.alvestrand.no/objectid/1.2.840.113549.1.1.11.html?fbclid=IwAR2wcEcMZ4YsLfhL00oByE1AHi6XtaBr5ezIlxCbKll3s5qYs5ewNRDnecI).

Graph API supports TLS 1.2 and 1.3 and non-static RSA cipher suites. We are currently deprecating support for older TLS versions and static RSA cipher suites. Version 16.0 no longer supports TLS versions older than 1.1 or static RSA cipher suites. This change will apply to all API versions on May 3, 2023.`

Is there any chance to get in running?

maakbaas, 2024-02-08:
Maybe its no longer relevant, or you found another solution. I recognize this can be an issue, but not immediately sure if and how this can be solved with BearSSL. I think the function calls you mentioned to try to set ciphers or SSL version is also what I found when shortly looking into this.

There is the website https://www.howsmyssl.com/ which you can send requests to and which then returns the information from your TLS request. 

At least in my case it is using TLS 1.2 and also seems to advertise some ciphers using SHA256, allthough I am not sure which one of these exactly maps onto the one Meta is requiring.

```json
{
    "given_cipher_suites": [
        "TLS_ECDHE_ECDSA_WITH_CHACHA20_POLY1305_SHA256",
        "TLS_ECDHE_RSA_WITH_CHACHA20_POLY1305_SHA256",
        "TLS_ECDHE_ECDSA_WITH_AES_128_GCM_SHA256",
        "TLS_ECDHE_RSA_WITH_AES_128_GCM_SHA256",
        "TLS_ECDHE_ECDSA_WITH_AES_256_GCM_SHA384",
        "TLS_ECDHE_RSA_WITH_AES_256_GCM_SHA384",
        "TLS_ECDHE_ECDSA_WITH_AES_128_CCM",
        "TLS_ECDHE_ECDSA_WITH_AES_256_CCM",
        "TLS_ECDHE_ECDSA_WITH_AES_128_CCM_8",
        "TLS_ECDHE_ECDSA_WITH_AES_256_CCM_8",
        "TLS_ECDHE_ECDSA_WITH_AES_128_CBC_SHA256",
        "TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA256",
        "TLS_ECDHE_ECDSA_WITH_AES_256_CBC_SHA384",
        "TLS_ECDHE_RSA_WITH_AES_256_CBC_SHA384",
        "TLS_ECDHE_ECDSA_WITH_AES_128_CBC_SHA",
        "TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA",
        "TLS_ECDHE_ECDSA_WITH_AES_256_CBC_SHA",
        "TLS_ECDHE_RSA_WITH_AES_256_CBC_SHA",
        "TLS_ECDH_ECDSA_WITH_AES_128_GCM_SHA256",
        "TLS_ECDH_RSA_WITH_AES_128_GCM_SHA256",
        "TLS_ECDH_ECDSA_WITH_AES_256_GCM_SHA384",
        "TLS_ECDH_RSA_WITH_AES_256_GCM_SHA384",
        "TLS_ECDH_ECDSA_WITH_AES_128_CBC_SHA256",
        "TLS_ECDH_RSA_WITH_AES_128_CBC_SHA256",
        "TLS_ECDH_ECDSA_WITH_AES_256_CBC_SHA384",
        "TLS_ECDH_RSA_WITH_AES_256_CBC_SHA384",
        "TLS_ECDH_ECDSA_WITH_AES_128_CBC_SHA",
        "TLS_ECDH_RSA_WITH_AES_128_CBC_SHA",
        "TLS_ECDH_ECDSA_WITH_AES_256_CBC_SHA",
        "TLS_ECDH_RSA_WITH_AES_256_CBC_SHA",
        "TLS_RSA_WITH_AES_128_GCM_SHA256",
        "TLS_RSA_WITH_AES_256_GCM_SHA384",
        "TLS_RSA_WITH_AES_128_CCM",
        "TLS_RSA_WITH_AES_256_CCM",
        "TLS_RSA_WITH_AES_128_CCM_8",
        "TLS_RSA_WITH_AES_256_CCM_8",
        "TLS_RSA_WITH_AES_128_CBC_SHA256",
        "TLS_RSA_WITH_AES_256_CBC_SHA256",
        "TLS_RSA_WITH_AES_128_CBC_SHA",
        "TLS_RSA_WITH_AES_256_CBC_SHA",
        "TLS_ECDHE_ECDSA_WITH_3DES_EDE_CBC_SHA",
        "TLS_ECDHE_RSA_WITH_3DES_EDE_CBC_SHA",
        "TLS_ECDH_ECDSA_WITH_3DES_EDE_CBC_SHA",
        "TLS_ECDH_RSA_WITH_3DES_EDE_CBC_SHA",
        "TLS_RSA_WITH_3DES_EDE_CBC_SHA"
    ],
    "ephemeral_keys_supported": true,
    "session_ticket_supported": false,
    "tls_compression_supported": false,
    "unknown_cipher_suite_supported": false,
    "beast_vuln": false,
    "able_to_detect_n_minus_one_splitting": false,
    "insecure_cipher_suites": {},
    "tls_version": "TLS 1.2",
    "rating": "Probably Okay"
}
```

Now, If I set the cipher using `httpsClient->setCiphers({BR_TLS_RSA_WITH_AES_256_CBC_SHA});`, I get the following response, so at least this function is working.

```json
{
    "given_cipher_suites": [
        "TLS_RSA_WITH_AES_256_CBC_SHA"
    ],
    "ephemeral_keys_supported": false,
    "session_ticket_supported": false,
    "tls_compression_supported": false,
    "unknown_cipher_suite_supported": false,
    "beast_vuln": false,
    "able_to_detect_n_minus_one_splitting": false,
    "insecure_cipher_suites": {},
    "tls_version": "TLS 1.2",
    "rating": "Improvable"
}
```

closed 2024-02-08 by maakbaas